aesim-tech/simba-project #27: Complementary PWM model

physicboy, 2020-04-24:
To provide a specific PWM with only single signal input:
- [ ] add dead time with rising edge delay mode

- [ ] add dead time with balance mode rising edge delay and advance falling edge

- [ ] add PWM duty cycle periodic sampling

aesim-tech, 2020-04-24:
Thanks, I didn't know it was useful to separate rising edge and falling edge delays. 
What do you mean by "PWM duty cycle periodic sampling" ?

physicboy, 2020-04-25:
[image]

though the rising edge delay is more popular, but if used in three phase grid inverter, you will see this pwm method will give more THD since it makes weird delay for the three pwm channel...

but, the second method will add balance delay...and the problem caused by the rising edge delay PWM...

sorry, I really dont know how to explain this...

anyway, for PWM duty cycle periodic sampling means that the duty cycle for the whole one whole PWM period is sampled when PWM counter is zero, or peak, or both zero and peak, so that there wont be any glitches in PWM output...

Maybe later I can provide the basic algorithm to implement this PWM dead time stuff + C code discrete sampling + and also PWM duty periodic sampling...later you may incorporate it in aesim in your best way

For better explanation, you can read the following article
[image]

[TMS320x2802x, 2803x Piccolo Enhanced Pulse Width Modulator (ePWM) Module.pdf](https://github.com/aesim-tech/aesim-project/files/4532376/TMS320x2802x.2803x.Piccolo.Enhanced.Pulse.Width.Modulator.ePWM.Module.pdf)