aesim-tech/simba-project #29: V-C loop, I-L Loops and floating nodes

aesim-tech, 2020-04-26:
### Background

It is not possible to connect two ideal voltage sources in parallel or two ideal current sources in series. For example, an ideal voltage source connected to an ideal capacitor will obviously lead to infinite current. In real-life, the impedance in the wire, source and capacitor will limit this current to a very large but finite value (until something eventually burns). 
In simulation, we don't have such impedance if the user doesn't include them in their model. Since we use ideal switch models, V-C and I-L loops are relatively hard to solve in AESIM and this type of errors and issues are usually thrown:
* "_Not able to solve the system (Singular matrix)_" error
* High-frequency numerical oscillation and "_Time step converged toward zero_" error
* Extremely slow simulation

Floating nodes generate similar issues for similar reasons.

### Solution

Even an experienced user can build a circuit with such conditions without knowing it. For example, A diode bridge grounded on one side  can be floating on the other side if all the diodes are opened. So, relying on the user's capacity to detect and correct floating nodes, V-C and I-L loops is not a good option.

Another option (which is done in other tools) is to automatically include large/small impedance to the source and switch devices. I have tried it and it solves most of the stability issues. However, there is a big performance drop (about 25%) due to the matrix getting more complex to invert.   

### Final Plan

As always in simulation, everything is a question of balance between simplicity, accuracy and performance. The following plan should solve all the stability issues without compromising performance and simplicity.

- [ ] Add a "Serie Resistance" parameter to Capacitor, Inductor, Voltage and Current Sources (DC, AC and Controlled)
- [ ] Detect V-C and I-L loop (not trivial because it must be done dynamically after every switch state change) 
- [ ] If V-C or I-L loop detected, throw a warning message and set the corresponding serie resistance parameter to a small value (1 mOhm?).
- [ ] Detect floating nodes (dynamically) 
- [ ] If floating node detected, throw a warning message and connect the node to the ground through a large resistance (1 MOhm?)

physicboy, 2020-04-28:
Make expert mode and basic mode simulation,

expert mode use all ideal component with benefit of faster simulation, but at risk of non converging...

Simulation in NL5 also said to use only ideal component without any hidden trick...

physicboy, 2020-04-28:
Anyway, if you decided to add series resistance, it should be in an order that is very small as such it is not practically used, maybe 1nano-Ohm ~ 1micro-Ohm

And for floating node, I think PSIM also throw error if there is floating node,

anyway when you said dynamic check for floating node, is it like in case of half bridge with dead time, but user use ideal bidirectional switch (without reverse conducting/body diode)?

aesim-tech, 2020-04-28:
> Make expert mode and basic mode simulation,
> 
> expert mode use all ideal component with benefit of faster simulation, but at risk of non converging...
> 
> Simulation in NL5 also said to use only ideal component without any hidden trick...

I really don't think it is required to have an "expert mode": If we are able to detect and solve all illegal conditions and warn the user about the changes we made, we should not have any problem. By default, all models are ideal unless specified by the user.

Anyway, I will try to implement this and then we can iterate and change until we find the perfect solution!

aesim-tech, 2020-09-03:
This was implemented for the first public beta release. Since it works well and do not impact simulation results, it is not required to throw a warning to the user.

closed 2020-09-03 by aesim-tech