aesim-tech/simba-project #58: Search feature

arharsha, 2020-10-05:
Ability to search through various library elements via a search bar or by a tool tip on the model page will help in decreasing model development time
[image]

aesim-tech, 2020-10-05:
Great Proposal Arharsha!

We have already an issue opened for the search feature (#23). I will close the other one and keep this one opened. 

The plan is to develop a unique search engine that searches in Libraries, Designs, Results, and Test Benches.
For instance, if you type "Resistor", it will show:
- The "Resistor" model in the library (to add a new "Resistor" model to the active circuit)
- Every instance of "Resistor" in the active circuit ("R1", "R2", "R3"...)
- Every chart with "Resistor" scopes
- ...

The idea is to have this search engine as reactive as possible with a nice UI you can invoke from anywhere with a shortcut. 

Let me know if you have any comments. This will be implemented in 4-6 months.

aesim-tech, 2021-10-04:
Similar feature developed: #160

closed 2021-10-04 by aesim-tech