aesim-tech/simba-project #121: save final states and load initial states

isablanca, 2021-03-23:
If the system has some slow dynamics, (e.g. The rotor of wind turbines with large inertia), we may not want to simulate the system for a very long time every time to reach the desired steady state. Instead, just like the Simulink software, if we can save the final states of simulation and load the initial states for the next run, we can study the system in a certain operation point more efficiently.

Besides, the save/load feature could be useful when calling the software from another simulation environment,  to perform the hybrid simulation. e.g. [1]

[1] D. Athaide, J. Qin and Y. Zou, "MATLAB/Simulink-Based Electromagnetic Transient-Transient Stability Hybrid Simulation for Electric Power Systems with Converter Interfaced Generation," 2019 IEEE Texas Power and Energy Conference (TPEC), College Station, TX, USA, 2019, pp. 1-6, doi: 10.1109/TPEC.2019.8662135.

aesim-tech, 2021-03-23:
100% agree. This feature is actually already developed and tested at the solver level. We just need to build a nice user interface around it.

aesim-tech, 2022-05-17:
Duplicate of #162

closed 2022-05-17 by aesim-tech