aesim-tech/simba-project #137: THD measurement

aesim-tech, 2021-04-29:
- [ ] Control measurement block
- [ ] Add option in the Result view to calculate the THD between two cursors

jerome-secretgit, 2023-05-16:
RMS Control measurement block

[image]

jerome-secretgit, 2023-05-16:
THD control measurement block

[image]

Control block
Parameters:

    Frequency of the fundamental in Hz (to get the period)
    Checkboxes to scope the output / input
    
    Contant window or floating window ?

[image]
Max considered harmonic:
- 50
- user value
- Use Parseval criteria

[image]

jerome-secretgit, 2023-05-16:
[image]

jcornau, 2023-06-08:
RMS device : symbol OK & solver OK --> to be validated
THD device : symbol OK
Power Probes : symbols OK

[image]

closed 2023-07-10 by manu-aesim