aesim-tech/simba-project #154: Support Thermal Data XML Format

aesim-tech, 2021-06-30:
- [ ] Option to create a new thermal data from a XML file (PLECS Format)
- [ ] Support Table-based data
- [ ] Support RC Thermal Network Model 
- [ ] Support Equation-based data

manu-aesim, 2022-09-08:
Game Plan: 

1. Support Thermal Dissipation Network
- [ ] Simba.Solvers
- [ ] Simba.Data.ThermalData
- [ ] Database
- [ ] Support Cauer and Foster Network Data

2. Load XML Data
- [ ] Add button "Create ThermalData from XML Description file"

3. Decide how to handle the variables
- [ ] When creating thermal data?
- [ ] At the thermal data level? 
- [ ] At the device level? 

4. Decide how to handle analytical expressions

manu-aesim, 2022-09-08:
[ThermalAnalysis_usecases.pptx](https://github.com/aesim-tech/simba-project/files/9530444/ThermalAnalysis_usecases.pptx)

manu-aesim, 2022-09-08:
```python

import xml.etree.ElementTree as et
import numpy as np
import pandas as pd
from pathlib import Path

def get_switching_energy(semiconductordata_xml_node, switch_loss_type, xmlns):

    switching_loss_xml_node = semiconductordata_xml_node.find(switch_loss_type, xmlns)
    switching_current = np.fromstring(switching_loss_xml_node.find('plecs:CurrentAxis', xmlns).text, sep=' ')
    switching_voltage = np.fromstring(switching_loss_xml_node.find('plecs:VoltageAxis', xmlns).text, sep=' ')
    switching_temperature = np.fromstring(switching_loss_xml_node.find('plecs:TemperatureAxis', xmlns).text, sep=' ')
    switching_loss_dict = {'plecs:TurnOnLoss': 'turn_on', 'plecs:TurnOffLoss':'turn_off'}
    
    energy_xml_node = switching_loss_xml_node.find('plecs:Energy', xmlns)

    for tJ, xml_node_temperature in zip(switching_temperature, energy_xml_node.iterfind('plecs:Temperature', xmlns)):
        for v, xml_node_voltage in zip(switching_voltage, xml_node_temperature.iterfind('plecs:Voltage', xmlns)):
            switching_energy = np.fromstring(xml_node_voltage.text, sep=' ')
            switching_energy *= float(energy_xml_node.get('scale', 1))
            df = pd.DataFrame(switching_energy, switching_current)
            df.to_csv(switching_loss_dict[switch_loss_type] + '_' + str(int(v)) + 'V_' + str(int(tJ))  + '.csv', header=False)    

reference = 'FF2MR12KM1_mosfet_simple'
xmlns = {'plecs': 'http://www.plexim.com/xml/semiconductors/'}
xml_tree = et.parse(Path(reference + '.xml'))
xml_root = xml_tree.getroot()
package_xml_node = xml_root.find('plecs:Package', xmlns)
semiconductordata_xml_node = package_xml_node.find('plecs:SemiconductorData', xmlns)

# Conduction Loss
conduction_loss_xml_node = semiconductordata_xml_node.find('plecs:ConductionLoss', xmlns)
conduction_current = np.fromstring(conduction_loss_xml_node.find('plecs:CurrentAxis', xmlns).text, sep=' ')
conduction_temperature = np.fromstring(conduction_loss_xml_node.find('plecs:TemperatureAxis', xmlns).text, sep=' ')

voltage_drop_xml_node = conduction_loss_xml_node.find('plecs:VoltageDrop', xmlns)

for tJ, temperature_xml_node in zip(conduction_temperature, voltage_drop_xml_node.iterfind('plecs:Temperature', xmlns)):
    voltage_drop = np.fromstring(temperature_xml_node.text, sep=' ')
    voltage_drop *= float(voltage_drop_xml_node.get('scale', 1))
    df = pd.DataFrame(voltage_drop, conduction_current)
    df.to_csv('conduction_' + str(int(tJ))  + '.csv', header=False)

# Turn-on Loss
get_switching_energy(semiconductordata_xml_node, 'plecs:TurnOnLoss', xmlns)

# Turn-off Loss
get_switching_energy(semiconductordata_xml_node, 'plecs:TurnOffLoss', xmlns)

```

manu-aesim, 2022-09-22:
To do : 
- [ ] Determine Function Variables list dynamically
- [ ] Update Custom Variable Table Header depending on Cauer or Foster
- [ ] Update Solver Thermal Impedance for Foster values
- [ ] Update U.I. (?)

closed 2022-10-20 by manu-aesim