aesim-tech/simba-project #159: Subcircuit Variables

aesim-tech, 2021-09-22:
- [ ] Add an option to define variables at the sub-circuit level. 
- [ ] Multiple instances of the same sub-circuit can have different variable values.
- [ ] If a sub-circuit cannot find a variable value in its definition, it will try its higher levels in the hierarchical order until it reaches the top circuit.
- [ ] If a subicircuit can't find a variable value in its definition or in its hierarchy, an error will be thrown.
- [ ] Subcircuit variables are accessible and modifiable via the Python API.

closed 2023-06-23 by manu-aesim

manu-aesim, 2023-07-27:
> For the first item "Add an option to define variables at the sub-circuit level" do you mean this?
> 
> --> consider to implement some kind of mask to access the inside parameters of the subsystem without the need of opening said subsystem. ??

Yes, if you look at the latest beta you can set variables on Subsystem and use those variables in the sub circuit content.

AdriPWS, 2023-07-27:
Ah yes, right !! thanks Manu ! (I just tested right now) !!