aesim-tech/simba-project #176: NullReferenceException: Object reference not set to an instance of an object

misso2020, 2022-02-11:
Hi, The issue I'm talking about happened :
1- When importing simba license  by the command : 
from aesim.simba import License
License.Activate(''*****************************'')

I got 1 as a result  instead of 3 
2- When I Tried to execute the following flyback converter  : 
# Load modules
from aesim.simba import DesignExamples
# Load project
flybackConverter = DesignExamples.DCDC_Flyback()
# Get the job object and solve the system
job = flybackConverter.TransientAnalysis.NewJob()
status = job.Run()
# Get results
t = job.TimePoints
Vout = job.GetSignalByName('R2 - Instantaneous Voltage').DataPoints

I got the following error:

---------------------------------------------------------------------------
NullReferenceException                    Traceback (most recent call last)
<ipython-input-2-539277ca2109> in <module>
      3 
      4 # Load project
----> 5 flybackConverter = DesignExamples.DCDC_Flyback()
      6 
      7 # Get the job object and solve the system

NullReferenceException: Object reference not set to an instance of an object
  at Simba.Data.DesignExamples.DCDC_Flyback () [0x000c5] in <6d6c7685de16470ea4ea0cb1898cb605>:0 
  at (wrapper managed-to-native) System.Reflection.MonoMethod.InternalInvoke(System.Reflection.MonoMethod,object,object[],System.Exception&)
  at System.Reflection.MonoMethod.Invoke (System.Object obj, System.Reflection.BindingFlags invokeAttr, System.Reflection.Binder binder, System.Object[] parameters, System.Globalization.CultureInfo culture) [0x0003b] in <7cd4deb8302a478d9d393f24754841da>:0 

I tried several attempts to solve this issue  but nothing works for me. I use anaconda python 7 , and Kubuntu linux as opration system
Any ideas to solve this problem?

[image]

aesim-tech, 2022-02-11:
Hello,

At the `License.Activate` Line, you need to include your Deployment Key (you will find it on your profile page). 

If you Deployment key is 123, you have to enter: 
`License.Activate("123")
`

I just tried on my side and I am not able to reproduce your error using a valid deployment key.  I will close this ticket for now but feel free to reopen it if the problem is not solved on your side.

closed 2022-02-11 by aesim-tech

misso2020, 2022-02-11:
Thanks, but i covered my deployment key in purpose, its not a license problem its problem of having null values somewhere in the script , I don't know how to solve it.

misso2020, 2022-02-12:
Thanks again, 
I sent my license to your email.
My aesim.simba package is 2021.12.2 and I have full access to the internet, I tried several times to make it work but no way, the interesting thing that even if I put anything in Activate.License(' ') , it outputs the same error.

[image]

Also when  I comment from aesim.simba import DesignExamples, I got the same message

[image]

Any suggestions please.

aesim-tech, 2022-02-20:
Hello,

I sent you an email last week. Have you received it? Is your problem solved? Unfortunately, I am not able to reproduce the bug.

misso2020, 2022-02-20:
Thanks so much Emmanuel , I received your email and I answered it, till now the issue couldn't be solved, I  uninstalled anaconda 3 entirely  with all its components, then I reinstall it but the problem is always there , I made a lot of search in the net , but no way. I hope that I can find some help in the near future.
Thanks

aesim-tech, 2022-02-21:
Messaoud,

Which Linux Distribution are you using? Is it a physical or virtual
machine? If it is virtual, where is it hosted (AWS... Google...)?
If you want to troubleshoot that issue with me, can you please Book a
meeting with me <https://meetings.hubspot.com/manu27>?

Thanks,
Emmanuel.

misso2020, 2022-02-22:
Dear Emmanuel,

Thanks a lot for your time and effort for solving issues related to simba software, I'm glad to inform you that this evening I solved my problem and started using aesim.simba package on  kubuntu 21.10 without problem.
Many thanks for your proposition of booking a meeting.

aesim-tech, 2022-02-23:
Hello Messaoud,

Thanks for sharing this good news! I'll close the ticket for now. 

Regards,
Emmanuel.