aesim-tech/simba-project #193: Propose a DFT instead of a FFT

guillaumefontes, 2022-07-27:
As a user, I would like a computation of the *Fourier spectrum* :
* option 1:
  - a *step frequency*,
  - a number of harmonics,
  - a number of analyzed periods (nb_periods).
 The time span (nb_periods x 1 / step_frequency) will be considered from end time.
* option 2:
  - tmin (or left cursor?),
  - tmax (or right cursor?),
  - a number of harmonics.

The Fourier coefficients are computed such as:
[image]

where:
[image]
for continuous signals.
and 
[image]
for discrete signals.
A piecewise linear approximation is used for continuous signals. This approximation accuracy depends on the size delta t

manu-aesim, 2023-01-24:
https://github.com/aesim-tech/simba-project-internal/discussions/118

jcornau, 2023-04-25:
GUI is now updated with the new menu (2 modes, frequency step / time range).
The time mode is synchronized with the cursors (a round at 10-6s is applied).
The FFT graph is now using only points (the "line" of the curve is disable).

guillaumefontes, 2023-04-29:
## Three issues according to me:

#### 1- horizon time considered
I do not know the horizon time which is considered for the FFT when I specify a frequency step + nb of harmonics + nb of analyzed periods. For example:
- simulation time of 0.06s
- (frequency step of 50 Hz + nb of analyzed periods = 1)
  => horizon time to be considered = 0.02s
Yet, I do not know if this horizon time of 0.02s has been considered in the beginning of the simulation, in the middle of in the end.

[image]

Solution = consider adding 2 horizontal cursors to show the horizon time which can be moved by the user

#### 2- Curve color
The curve color is not the same in the time chart and in the frequency chart for a considered variable. See the picture above: the voltage waveform is in yellow but the frequency chart is in blue.

#### 3- Zoom out
When I first zoom in and then zoom out, the frequency spectrum disapp
[image]
ears...

guillaumefontes, 2023-05-01:
## New bugs found:

### Bug with scale / magnitude
Example: these 2 signals have different magnitudes but they show the same scale / magnitude for the frequency spectrum
[image]

### Accuracy of computed magnitudes
Example with a square wave between 0 and 1 with a switching frequency of 1kHz and a duty cycle of 0.5.
The dc component shows a value which is slightly below 0.5
The 1st harmonic shows a magnitude of 0.647 and it should be 0.637.
(and it doesn't seem to depend on the number of harmonics)
[image]

### Aliasing pbm ?
Same example but with a number of harmonics = 100
[image]