aesim-tech/simba-project #241: Reduce the number of signal points

manu-aesim, 2023-02-15:
**Is your feature request related to a problem? Please describe.**

* Between two discontinuity events, SIMBA always calculates at least 4 points to ensure that we simulate all time constants.  These points are saved in the scopes but may not carry any additional information. 

**Describe your desired solution**.
We need to consider whether we can remove these points from the scopes. A simple option would be that if the LTE (local truncation error) calculated on these points is low (1E-8?), to remove these points from the scopes.

jcornau, 2024-04-19:
Performances 
|                        | Pas de compression | Compression à 1e-5 | écart |
|------------------------|--------------------|---------------------|-------|
| Fixed time step        |                    |                     |       |
| Nbre de points         | 21 233 721         | 3 845 254           | -82%  |
| Temps                  | 15132 ms           | 16816 ms            | 11%   |
| Variable time step     |                    |                     |       |
| Nbre de points         | 5 152 435          | 1 617 828           | -69%  |
| Temps                  | 4131 ms            | 4290 ms             | 4%    |

manu-aesim, 2024-04-19:
- [ ] Update Doc

manu-aesim, 2024-04-22:
Preview version 

https://aesimtech.s3.us-east-1.amazonaws.com/artifacts/R2planF4ytgohfyH0944X3VL8gBoYjB3/simba.exe
pip install aesim.simba==2024.4.22a1

manu-aesim, 2024-04-23:
Scope compression is a feature designed to optimize memory usage. This feature is particularly useful in scenarios where simulations generate a large number of points. This option is available in the simulation settings and is disabled by default.

[image]

When a simulation is running, Simba solver will apply an algorithm to calculate the variation between consecutive points. If the relative variation is not significant (relative error lower than 1e-5), the point will not be saved in the scopes. Activate this option can drastically reduce the memory size of the simulation and the time to load the simulation results in the GUI.

**Note :** 
Sampled signals (when "Sampling time" is defined) are not affected by the compression.

[image]

In yellow : no compression
In blue : with compression

closed 2024-04-23 by manu-aesim