aesim-tech/simba-project #246: Fourier analysis for AC sweep replaced with discrete time fourier series

physicboy, 2023-02-24:
Discrete time fourier series is completely memory-less.

Compared to FFT, DTFS is much slower if we want to compute the spectrum of a signal. But for AC sweep, where we are only interested in 1 frequency, DTFS can be more suitable and can be beneficial for very low frequency sweep.

manu-aesim, 2023-02-24:
Hi,

The AC Sweep is computed in two phases.

**1. Steady State Detection.**
In this phase, we do not apply any perturbation and we look for the stationary state of the system. Indeed, we use an FFT algorithm. To determine if the steady state is reached, SIMBA analyzes the RMS value and the highest non-DC harmonics (if any) of all simulated state variables. We must use an FFT at this point because the frequencies of the variables are unknown. 

**2. AC Sweep**
When the steady-state of the system is reached. we copy the state of the solver and duplicate it on different threads (one per frequency we are interested to simulate). We apply the perturbation and use a DFT algorithm to calculate the transfer function between the output and the perturbation.  So we actually already run a DFT for the AC Sweep at each frequency. We are actually using the Goertzel Algorithm (https://en.wikipedia.org/wiki/Goertzel_algorithm). 

Please reopen this ticket if this is not clear or if you want to continue the discussion further.

physicboy, 2023-03-02:
Hi Manu,

My point was to implement the algorithm in memory-less approach. As I notice issues with memory handling (especially on transient simulation). But in a glance that Goertzel algorithm appears to be very similar with DTFS that I meant

manu-aesim, 2023-03-06:
Hi,

We're aware of memory issues when running large simulation over a long period of time with small time steps. This is why we have opened [this issue ](https://github.com/aesim-tech/simba-project/issues/234). This is not only related to AC Sweep. Let us know if anything.

closed 2023-03-06 by manu-aesim