aesim-tech/simba-project #247: Numerical instability with MOSFET+anti parallel diode+Cds+deadtime

physicboy, 2023-03-02:
As mentioned in the email, I noticed severe numerical oscillation on simulation result where the MOSFET is connected with anti parallel diode+Cds and deadtime circuit.

The problem is resolved when anti parallel diode, Cds, and deadtime is removed.

guillaumefontes, 2023-03-03:
Thanks for reporting this!
This may be linked with an issue of the diode switch-off. We are looking into it.

manu-aesim, 2023-03-06:
Yes, we looked into it this morning with Guillaume. We found a bug when multiple diode zero-switching interpolations occur in a single time step with the predictive time step. We have a version that fixes this issue but we need a few days to validate and test. We will send you a beta version in a few days to check if it solves your issue. 

Thanks for your patience!

physicboy, 2023-03-13:
Hi Manu/Guillaume

Sounds good...thanks

manu-aesim, 2023-03-21:
Hi @physicboy,

Just a small update on this to let you know that I am still working on this. To be honest, it is more complex that I thought and I am still working on a fix. I hope to have it this week.

I thank you for your patience.

physicboy, 2023-03-22:
Hi @manu-aesim

Got it,

I can imagine it to be not a simple issue.

Keep up the good work...

manu-aesim, 2023-03-25:
Hi Arief,

We have released a new beta version of SIMBA, which you can download here: https://aesimtech.s3.amazonaws.com/artifacts/k9kdJDCyXjjzgHAw9gGuXeM7JJL3e3ph/simba.exe

We have made several improvements to enhance the stability and accuracy of the solver, although it may still not be perfect. Here is a summary of the changes we've made:

1. We have completely recoded the algorithm that interpolates the simulation at the time of zero crossing for diode voltage and current. There was a conflict in the latest release with the Optimal Time Step Algorithm, which has now been resolved.
2. We've addressed the issue of numerical noise that occurs when the time step is small or changes abruptly. This noise can cause small voltage/current variations, especially in less well-conditioned (floating) parts of the circuit. Although this noise is very small (in the 1E-9 range), it could cause issues with diodes entering an unstable state (randomly opening or closing), leading to spikes and inaccuracies. We have improved this by being more strict in detecting when a diode state must change.

While we are aware the accuracy of the model simulation is not 100% perfect yet, we would greatly appreciate your feedback on the quality of the results. Please let us know if you encounter any issues or have further suggestions for improvement, I will close the issue for now, but feel free to reopen it. 

Thanks,
Manu.

closed 2023-03-25 by manu-aesim

physicboy, 2023-03-27:
Hi Manu,

Ok. I got it and will try it.

By the way, could you consider to add the file version to the file name (simba_2_1_0.exe)? So it will be easier for me to know which one is the latest version as I downloaded so many versions of simba.

manu-aesim, 2023-03-28:
Hi Arief,
It's a good idea but I have to update our CI code to make it happen. I will see if I can do it for our next official release.