aesim-tech/simba-project #255: Discontinuity Event in C Code blocks

manu-aesim, 2023-04-07:
**Is your feature request related to a problem? Please describe.**
Capability to plan and raise discontinuity events in C Code. This can be useful for any code generating PWM signals like Space Vector Modulation or Discontinuous PWM.

**Describe the solution you'd like**
The C Code block:

-  should have a new primitive named "nextDiscontinuityEvent()" where the user can calculate when the next discontinuity event will happen.
- should have a way to raise a discontinuity event when the output changes. The user can call "discontinuity_event()" in the calculate_output() function.

physicboy, 2023-04-10:
Hi Manu,

I am curious to know how discontinuity event can improve PWM generation?

I can understand if it can improve the solver efficiency, because the solver can know exactly the switching event without needing to detect discontinuity and goes back and forth to accurately pinpoint the exact discontinuity event (that's caused by the PWM)

manu-aesim, 2023-04-12:
@physicboy,
Indeed, that is the current limitation. As of now, we cannot force the solver to establish a specific time point through the use of C-code blocks. For example, when developing a Space Vector Modulation utilizing the C-Code block, we encounter this constraint.

physicboy, 2023-04-14:
@manu-aesim,

I think that can significantly improve my circuit diagram and also make it more efficient... as you may remember I have quite a lot of PWM... (though not as many as multilevel things)

* I remember I asked you if you would be interested to pre-calculate the PWM switching point for every switching period....haha

Would love to do beta testing on this feature...

manu-aesim, 2023-04-14:
@physicboy, Good to know :)
I hope to release this feature by the end of the year.

In the meantime, have you tried using the multi-time step solver on the C code? This will force the C code to be called at a given rate (like a microcontroller) and should not cause problems with discontinuity events. If this is not working, please let me know.

physicboy, 2023-04-20:
Yes I tried that multi-step solver, but because I also need to perform some extra measurement sampling in between the control algorithm sampling rate, at the end I used the older trigger block

I think this approach for discontinuity event will be more suitable as it allows me to do multiple trigger inside the C-block

physicboy, 2023-10-19:
Hi @manu-aesim 

I am now working with Qspice, you may find this example here to be relatable to your quest....
https://forum.qorvo.com/t/interesting-case-of-c-code-with-trunc/15752/2

manu-aesim, 2023-10-19:
> Reference in new issue

Hi Arief, 
Hope you are doing well. Thanks for sharing this.  If my understanding is correct, the Qspice DLL can control the size of the next time step. What I don't understand is it should be able to control the size of the previous time step( the one before the discontinuity) to ensure that the discontinuity time accuracy is lower than TTOL. Am I missing something ? 

What I envision in SIMBA is to add one function with this signature to our C Code interface : 

``` C 
double next_discontinuity_event(void current_time){...}
// The user can specify when the next discontinuity event will happen.
// SIMBA will add one time point right before the discontinuity to ensure the discontinuity time is simulated with high accuracy. 

```

and the capability to specify if a discontinuity happened in the calculate_outputs function. 

``` C 
void calculate_outputs(double* outputs, double* inputs, double time, double time_step, bool& notify_discontinuity_event) {
// Set notify_discontinuity_event  to true if a discontinuity happened. 
}
```

physicboy, 2023-10-20:
Hi Manu,

In my limited understanding, the way Qspice may work is by using a logarithmically increasing time step. However upon detecting a discontinuity event (or large deviation of expected signal trajectory). When it happened, then it will discard the solution at t(n), revert back to t(n-1) and start again with small time step for a few steps (defined as TTOL), then continue with again the logarithmically increasing time step. You may see this from the picture below.

[image]

The thing about their C-code is, by calling the Trunc() function, the C-code can trigger the discontinuity detection and forced the overall solver to revert back t(n-1) and start again with small time step.

However, one potential difference between Simba solver and Qspice solver is, Qspice solver deltaT is always positive. 

I think your proposed solution should work well and potentially faster compared to Qspice. But, the application may be more limited to predefined next event (i.e. PWM generation from discrete controller). While for Qspice their solution can also allow the C-code to accurately track the discontinuity caused by the event from analog circuit. (I implemented peak current mode control there, there is no way the C-code can know when the switch turn off event should happened).