aesim-tech/simba-project #274: Implement autosave/auto backup

physicboy, 2023-05-25:
Hi Simba Teams,

I often use Simba in edge cases, and it results in the software just crashing and closed on its own.

I understand that improving the software may take time, but in the meantime, is it possible to implement autosave/backup save every predetermined time interval so that in case the software crash we dont lose the last hour work that accidentally hasnt been saved.

Thanks,
Arief,

manu-aesim, 2023-05-26:
Hi Arief,

I initially had reservations, but since many users have requested it, I've decided that it's time to implement an autosave feature that saves data to a temporary backup file. Here's the proposed functionality:

- While SIMBA is in use, the ongoing project will be autosaved every minute or two to a temporary backup file, given there have been changes.
- This backup file will be deleted upon the proper shutdown of SIMBA (i.e., when SIMBA doesn't crash).
- If, upon startup, SIMBA finds any existing temporary backup files, a message will be displayed: "SIMBA did not close properly during the last session, would you like to restore unsaved projects?"

I would like to hear your thoughts on this. Any feedback or potential improvements to this feature would be appreciated.

physicboy, 2023-05-26:
You can do it in Ms. office style which is exactly the same as you described above

Or do it in the same style as some CAD software by making automatically making a backup folder in the same location of the working file
i.e. file is at d:\work\simba\pfc.simba
Then you automatically create a backup folder (if not yet existed)
d:\work\simba\~pfc_backup
Then just dump all the backup file there.
You can use date and time suffix for the file name
d:\work\simba\pfc_20230526_2319.simba

Of course the backup folder size will grow a lot, but I think it can be a good time machine...and if user dont need it, they can delete it themself manually (or you can set max backup folder size or max file age like days/weeks)

About the frequency of auto backup, You can have a few options:
1. Every certain fix time duration
2. Every N activities (add/remove/move components/wire)
3. Everytime you close the C code editor
4. Everytime you press run simulation.

For me I notice I always have crashing issue when  I simulate my model (so, I think option 4 is good enough)

physicboy, 2023-05-26:
You can do it in Ms. office style which is exactly the same as you described above

Or do it in the same style as some CAD software by making automatically making a backup folder in the same location of the working file
i.e. file is at d:\work\simba\pfc.simba
Then you automatically create a backup folder (if not yet existed)
d:\work\simba\pfc~backup
Then just dump all the backup file there.
You can use date and time suffix for the file name
d:\work\simba\pfc~backup\pfc_20230526_2319.simba

Of course the backup folder size will grow a lot, but I think it can be a good time machine...and if user dont need it, they can delete it themself manually (or you can set max backup folder size or max file age like days/weeks)

About the frequency of auto backup, You can have a few options:
1. Every certain fix time duration
2. Every N activities (add/remove/move components/wire)
3. Everytime you close the C code editor
4. Everytime you press run simulation.

For me I notice I always have crashing issue when  I simulate my model (so, I think option 4 is good enough)

manu-aesim, 2023-05-26:
Hey Arief,

I actually developed this this morning: https://aesimtech.s3.amazonaws.com/artifacts/mYSE8UkPSmoeqXW5B9lBNMPom6a7jvrM/simba.exe

It's actually quite difficult to detect whether SIMBA crashed or not during the previous session so I've implemented this simple logic: 
- Every minute, if the design has changed, the project file is saved silently in a back-up folder. There is one backup file per project file. 
- There is a new option in the Welcome View to open this folder and check the files that can be restored.

[image]

Let me know what you think.

On another note, I have removed the support of .simba files in the next version of SIMBA : https://github.com/aesim-tech/simba-project/issues/277. If you are still using .simba files, you should convert them to .jsimba files.

physicboy, 2023-05-27:
I dont think if you actually need to detect when crash happened, and when the software crash i dont think if you can reliably detect how far the crash occurs...

What you do (backup every minute) is the most foolproof method. Additionally, i still strongly suggest you to add backup everytime you press the "Run Simulation", which should be foolproof and also when I got most if not all crash.

About the button placement to open backup folder, I think you can make it more obvious

physicboy, 2023-05-29:
By the way, do you consider do auto backup even to files that has never been saved?

manu-aesim, 2023-05-29:
> By the way, do you consider do auto backup even to files that has never been saved?

I have explicitly disabled this. Do you think that would be useful?

physicboy, 2023-05-29:
I think it 

> > By the way, do you consider do auto backup even to files that has never been saved?
> 
> I have explicitly disabled this. Do you think that would be useful?

Yes, it will be helpful,,,, pretty sure if I am not the only one that can crash the software from a quick simple design . Though,i dont know where you want to keep the backup file

closed 2023-06-23 by manu-aesim