aesim-tech/simba-project #279: C Code Optional Additional Include Directory

manu-aesim, 2023-05-30:


manu-aesim, 2023-05-30:
Beta version: https://aesimtech.s3.amazonaws.com/artifacts/zZnoYa5u3aBEBNWALW97W29WPyzdQGHf/simba.exe

physicboy, 2023-05-31:
Hi Manu,

Do you consider to implement this function to use simba to call the compiler for all C-code/header file.
Perhaps by providing the directory of the code files and the main file

I think for me it will be a little confusing if one file is embedded inside Simba file and other files are in separate folder

manu-aesim, 2023-05-31:
> Hi Manu,
>
> Do you consider to implement this function to use simba to call the compiler for all C-code/header file. Perhaps by providing the directory of the code files and the main file
>
> I think for [...]

Hi Arief,
Not sure to understand. You can also use the "C Code" External File to have all the C Code into the same directory and nothing in SIMBA.

closed 2023-08-14 by manu-aesim