aesim-tech/simba-project #286: Universal Motor Request

vishal240294, 2023-06-07:
Dear Team,
An OEM ATOMBERG is looking for Universal motor for grinder application.

vishal240294, 2023-08-03:
@imaryoko can we get any update?

vishal240294, 2023-09-06:
@imaryoko Today customer meet me, can you please update something on roadmap for universal motor design in SIMBA.

imaryoko, 2023-09-07:
Hi Vishal, thanks for your request. At this point, this item's priority is after 2023 November release. To increase the priority, please put more information like business importance, how serious the customer is, or is it casual request, etc.

vishal240294, 2023-09-07:
Dear Ryoko, this is a very important prospect, and yesterday in person I meet the customer marketing manager. He said we need it soon as it is our upcoming prospect.