aesim-tech/simba-project #291: Connector spacing behavior on C-block

physicboy, 2023-06-20:
The c-block connector spacing changing is in non consistent manner.

I think if its better to keep the connector space to be consistent (2 grids for example).

[image]

manu-aesim, 2023-06-20:
Hi Arief,

I did that on purpose to keep the connectors symmetric. One thing that we could do is to align the top connectors and keep the same spacing. This is quite easy to implement but the problem is backward compatibility (we don't want existing design to be corrupted by this change). 

I prefer to wait until we have the symbol editor (end of 2023), where you will be able to place the pins wherever you want: https://github.com/aesim-tech/simba-project/issues/271

Let me know if you have a better option in mind :) I will close the ticket for now.

closed 2023-06-20 by manu-aesim

physicboy, 2023-06-21:
Hi Manu,

Well noted... 

Thanks