aesim-tech/simba-project #292: Bizzare C-code behavior

physicboy, 2023-06-21:
Hi teams,

I today experience sudden crash, and the C-code suddenly replaced with default C-code template after re-opening the simulation file

I dont know how to replicate the issue, but it just happened

Thanks,
Arief,

manu-aesim, 2023-06-22:
Hi Arief, 

Have you checked the backup file if it contained the C Code ? 

I will have a look. to this but it is hard for me to debug if we can't reproduce the issue.

Thanks,
Manu.

physicboy, 2023-06-26:
Hi Manu,

I cant find any file in the backup folder.

I will send you the simulation model in question via email.

Thanks,
Arief,

manu-aesim, 2023-07-10:
I will close that issue for now but let me know if it happens again please.