aesim-tech/simba-project #293: Minimum Timestep causes distortion

physicboy, 2023-06-22:
Hi Teams,

I realized that the minimum timestep parameters when using predictive solver can causes distortion under two cases:

1. Extreme low or high duty cycle
Assuming 5% duty or 95% duty on 25kHz PWM so the distance between on and off is 2us. In this case 1us minimum time step can distort the duty cycle quite significantly 
2. Multiphase converter
After switching event of one phase, it is possible that the minimum timestep to interfere with switching time of the other phase if it happened in close time proximity

Thanks
Arief

manu-aesim, 2023-06-22:
Hi Arief,

1. This is not really a bug but expected behaviour since minimum time step is also the accuracy of discontinuity events. If the period is 2us the min time step can't be equal to 1us since you can have an inacuracy of about 2*min_time_step (one at the beginning and one other at the end of the period). In this case, I recommend to have a min time step below than 0.1us. 

2. It is hard to assist you without more info. Is there a design you would like to share to illustrate this ? 

Thanks,
Manu.

physicboy, 2023-06-26:
Hi Manu,
1. Unfortunately, lowering the minimum time step is not always desirable... for example for losses analysis where I need to run the simulation for long total time to ensure steady state but I also need to ensure accurate simulation waveform.

2. Please check my the sketch in the attachment to illustrate my point.

[Potential conditions that can violate the minimum timestep.docx](https://github.com/aesim-tech/simba-project/files/11866092/Potential.conditions.that.can.violate.the.minimum.timestep.docx)

Thanks,
Arief,

physicboy, 2023-06-27:
Anyway, please note that for a three phase inverter as in the example 2, the resulting distortion should be not too significant because the converter will only be in that condition for 2-3 pwm periods

But, for interleaved DC/DC the situation can become quite severe as the converter will be in the particular duty cycle continuously

physicboy, 2023-06-28:
Hi Manu,

well noted. looking forward to that implementation.

as always, I will be open to do beta testing for this new solver implementation

Arief,