aesim-tech/simba-project #311: Bug: multi labels connected to the same connector

AdriPWS, 2023-07-20:
**Describe the bug**
There is a bug when connecting one block to multiple labels
see attached the error explaining clearly the issue

** SIMBA Version**
 - Version : 23.05

The issue is solved if you connect only one block to only one label.

[image]

manu-aesim, 2023-07-24:
Hello,

This is is not a bug but a limitation to avoid signal type conflict. I will close it for now.

AdriPWS, 2023-07-24:
Hello,

Thanks for this answer. So it seems right now we cannot connect one block to multiple labels.
However is there a workaround to realize this ?
Or we can only use one block for one label ?

Thanks,
KR

manu-aesim, 2023-07-24:
Hi,

If it's a control signal you can add a gain of 1 and connect its output to the new label.

closed 2023-07-24 by manu-aesim