aesim-tech/simba-project #317: Cursor line width

AdriPWS, 2023-07-25:
In the "result" tab, it would be good to have the capability to play with the width of Cursor line.
It is too thick.

manu-aesim, 2023-07-26:
I never saw a case where the width of the cursor were too thin or too thick. Would it be possible to share a screenshot of such a case? In any case, we will recode entirely the chart engine soon.

AdriPWS, 2023-07-26:
I don't have a case showing this but this is only the feedback of one user:  • Cursor line too thick. 
So I was wondering if a new "parameter" window could be added in "result" tab in order to play with the width of the cursor (see attached) and font/size of label (vertical axis)
[image]

closed 2023-07-26 by manu-aesim

AdriPWS, 2023-07-26:
"more customization"  but I do agree with you.