aesim-tech/simba-project #322: Simba with offline license, needs to regenerate machine ID and download new license after simba crash

physicboy, 2023-07-31:
same as the title
Simba with offline license, needs to regenerate machine ID and download new license after simba crash

another thing is, quite often randomly I need to download new license even before my monthly license expired

Thanks
Arief,

manu-aesim, 2023-08-02:
Hi @jpellegrinopws ,

Would you mind looking at the licensing code to understand why the offline license behaves like this for Arief ? 
We should not discuss the details of the licensing system here. 

Thanks,
Manu.

manu-aesim, 2023-11-09:
We are not able to reproduce this bug but we are in the process of rewriting our licensing system and that should solve this issue with offline activation.

closed 2023-11-09 by manu-aesim