aesim-tech/simba-project #324: Incorrect result of a thyristor bridge with a star/star transformer using the multi-time-step solver

jcornau, 2023-08-08:
**Describe the bug**
A simulation with a thyristor bridge and a star/star transformer is giving wrong results at specific time points when using the multi-time-step solver. 

It seams some commutations of the diods are not expected. The differences can also vary when changing the circuit (using a different grid frequency) or the simulation settings (different minimal time step).

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Language: Any

** SIMBA Version**
 - Version 2023.08 & 2023.05

manu-aesim, 2023-08-14:
@jcornau , I think it is now fixed. Can you please double check the original file causing the issue with the latest beta release ?

jcornau, 2023-08-16:
I checked with the original model and the problem is fixed in both cases:
- predictive time step and Sampling time activated,
- fixed time step solver.

closed 2023-08-18 by manu-aesim