aesim-tech/simba-project #326: waveform postprocessing

physicboy, 2023-08-09:
Hi Manu/All,

Please consider waveform to add capability for simple post processing (*/+-).

Main benefit is, it makes it easy to evaluate component power loss of passive component (RL, RC, or just R), I can probe the voltage and current.
Then use the post processing to get the power by V*I

Or sometimes, I need a differential voltage between two nodes and also their absolute voltage.
In this case I can do plot(waveA), plot(waveB), plot(waveA-waveB)

Thanks
Arief

manu-aesim, 2023-08-10:
Hi Arief,

I've seen this request many times and understand the benefit. The decision we've taken to meet this need is to add the ability for the user to write pre/post-processing python scripts directly in SIMBA.

This will give much more flexibility than a simple waveform calculator. We understand that some users will be afraid to write code, so we'll also be providing templates to help them.  

Will this work for you ?

physicboy, 2023-08-11:
I understand if the goal is to perform complex math, Python will be the better way to go.

I did quite a few complex waveform post processing in the past in Matlab, so I dont have much issue with doing it in Python.

But, we also do a lot of quick simple math to see how things correlate. And writing Python script, if possible, I hope can be avoided.

manu-aesim, 2023-08-11:
Hmmm... I think I agree with you. We could create a new type of job capable of processing waveforms and a simple interface to configure it.  
It's more complex than it sounds and we have to think about the following corner cases: What happens when two signals don't have the same sampling time? What happens when the start or end times of the waveforms are different? The best solution is probably to take the first signal as a reference and interpolate the other signals used in the equation to match the time points of the reference signal.

I will plan this for 2024.

physicboy, 2023-08-12:
Thanks for your effort.

In a glance, I thought it was a bit easier... LoL,

1. Just take the latest start time and earliest end time
2. Create an interpolation point when any of the source has hit a sampling time. It is supposed to be the consequences of variable time step, right?

manu-aesim, 2025-03-06:
Signal Calculator available in SIMBA 25.02.

closed 2025-03-06 by manu-aesim