aesim-tech/simba-project #327: Control PWL sources being compromised by connector labels

rcmartins72, 2023-08-10:
**Describe the bug**
Connector labels sometimes seem to disrupt the PWL source they are connected to - the connectors still seem to do their job of tying nets "over the air". The problem is that weirdly the source to which they were attached "forgest" its custom waveform.
I had initially a set of 5 PWL control sources each individually configured to a specific TimeValueMatrix using Python
All had connector labels connected to them.
If they only have connector labels connected to them everything works fine. However, when those connector labels then go on to connect to other control blocks that implement other operations then, for some reason, and rather randomly (meaning, not quite figured out the mechanism) the original PWL defaults to the original waveform. This seems to happen when the PWL are connected to a control-product operator. If instead of connecting through the labels I connect with "wires" then everything seems to work fine again, but obviously the legibility brought in by the labels is lost and the GUI becomes closer to the, to me dreadfull, LabView, kind of "coding".

**To Reproduce**
Steps to reproduce the behavior:
1. Create PWL sources (control) and replace their TimeValueMatrix with a matrix (numpy to list)
2. Connect two of these to a math product operator
3. Create a few of these blocks and plot the PWL output
4. See error

**Expected behavior**
The PWL that have not been connected to a product operator will still show the custom waveform. The ones connected to the product will have lost their custom waveform and defaulted to the original one.
[image]

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS :Windows 10
 - Language: Python

** SIMBA Version**
 - Version 23.5.5.0

manu-aesim, 2023-08-10:
Hello,

Would you mind sharing the python script causing the issue and/or the resulting simba project file that is not working properly ? 
Without this it is hard to understand exactly what is going on.  I've seen similar problems caused by label names: to work properly, label names must be perfectly identical (with no extra spaces and the same case). 

If it contains confidential data, you can mail it to me at manu@aesim.tech.

manu-aesim, 2023-08-14:
Problem solved outside of Github.