aesim-tech/simba-project #340: Python API does not work with 23.08

ArmiSchm, 2023-09-28:
I have observed two issue:

- A python-script, was running perfectly with a simba-project modified and stored in 23.05
- I have loaded this project into 23.08 and saved it.
- When I start the python-script observe two different types of errors:
1. project is loaded, but the Designs cannot been found with "GetDesignByName" (Hint: I've looked into the project data structure and see, that a) not all designs are imported. b) the Designname is not imported)
2. project cannot loaded. I'm receiving a JsonException: The JSON value could not be converted to System.Double. Path: $.Designs[0].TransientAnalysis.EndTime | LineNumber: 2077 | BytePositionInLine: 25.
 ---> System.InvalidOperationException: Cannot get the value of a token type 'String' as a number.
   at System.Text.Json.ThrowHelper.ThrowInvalidOperationException_ExpectedNumber(JsonTokenType tokenType)
   at System.Text.Json.Utf8JsonReader.TryGetDouble(Double& value)
   at System.Text.Json.Utf8JsonReader.GetDouble()
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value)
   at System.Text.Json.Serialization.Metadata.JsonPropertyInfo`1.ReadJsonAndSetMember(Object obj, ReadStack& state, Utf8JsonReader& reader)
   at System.Text.Json.Serialization.Converters.ObjectDefaultConverter`1.OnTryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value)
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value)
   at System.Text.Json.Serialization.Metadata.JsonPropertyInfo`1.ReadJsonAndSetMember(Object obj, ReadStack& state, Utf8JsonReader& reader)
   at System.Text.Json.Serialization.Converters.ObjectDefaultConverter`1.OnTryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value)
   at System.Text.Json.Serialization.JsonConverter`1.TryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, T& value)
   at System.Text.Json.Serialization.JsonCollectionConverter`2.OnTryRead(Utf8JsonReader& reader, Type typeToConvert, JsonSerializerOptions options, ReadStack& state, TCollection& value)

manu-aesim, 2023-09-28:
@ArmiSchm , thanks for the issue. We will look into it right away. Can you please confirm which version of the python module are you using (`pip show aesim.simba`)?

@AdriPWS, are you able to reproduce this issue ? If yes, can you please share a project file for 23.05 and a script causing those exceptions ? Thanks.

AdriPWS, 2023-09-29:
Hi, 
Please could you give me the version of the python module that you are using as asked by Emmanuel?
In command prompt, just type : pip show aesim.simba

Then I tried to reproduce your issue but I failed (sorry for that):

I downloaded SIMBA.exe version 23.5.5.0 and did a simple buck converter + saved the project (.jsimba file)
I downloaded aesim.simba version 2023.5.19 and run this simple buck converter in python and it worked well.

Now I downloaded aesim.simba version 2023.8.1a1 and launched the same script and it worked as well
Now I downloaded aesim.simba version 2023.8.31 and launched the same script and it worked as well

So I also loaded the .jsimba project created earlier into SIMBA.exe 23.08.31 and launched the same script
as before with still aesim.simba version 2023.8.31 and it worked again.

Sorry I don't succeed to reproduce your behavior.
Did you think to update the python version? pip install aesim.simba==2023.9.25a1
Can you try also please?
We can set up a meeting as well.

Thanks,
Adrien

ArmiSchm, 2023-09-29:
aesim.simba Version: 2023.5.19

ArmiSchm, 2023-09-29:
[InverterSim.zip](https://github.com/aesim-tech/simba-project/files/12756319/InverterSim.zip)

Find attached the necessary files to reproduce.

ArmiSchm, 2023-09-29:
Hi, I just updated to pip install aesim.simba==2023.9.25a1
and the error did not occur. 
THX

AdriPWS, 2023-09-29:
Hi,
Ok perfect, thanks for your positive feedback.
So everything is solved now.

ArmiSchm, 2023-09-29:
Yes. Perfect. 
I suggest to add some kind of revision check between the Simba.exe and the API. 

You can now close the topic. THX.

AdriPWS, 2023-09-29:
I totally agree with you and we already do that but I have tried to reproduce your issue without success.
I used 2023.9.25a1, 2023.8.31 and 2023.5.19 and even with those version I don't have your error message.
So I don't understand what's happened.

As your problem is fixed, yes we can close. But for sure we will double check that for later revision.

Thanks,