aesim-tech/simba-project #346: bug/crash with C block !

AdriPWS, 2023-10-09:
**Describe the bug**
I use the SIMBA 23.9 version for describing the following bugs and crash:
[Monostable_moyenne.zip](https://github.com/aesim-tech/simba-project/files/12844025/Monostable_moyenne.zip)
See attached the "zip" folder to reproduce the 2 issues:

If you open the zip folder and run the .jsimba file, you'll obtain 2 designs: "monostable" &  "moyenne".

Issue 1:
1) For "monostable" design:
run the circuit and the 2 outputs for each C block don't provide the same results.
Of course the C code is the same.
You need to put a "sample time" value inside the " C block (external)" to make it run with same results.
see the gif: "C_code_bug"
[image]

2 For "moyenne" design:
The bug/crash could be reproduced by following the "gif": "c_code_bug5".
[image]

In deed, SIMBA crashes when the 2 C blocks are into the same design for this case.
If you delete the "external one" or "normal", crash is avoided but it may happen randomly. And finally results check will provide a crash.

jcornau, 2023-10-12:
Issue 1 is fixed. 
External C code and internal c code are compiled differently. Also in this C code, the variable "out" is not initialized. This created a different behavior between both cases. Adding an initialization (out = 0) makes both c code block behaviors identical.

manu-aesim, 2023-10-18:
The crash is fixed in this release:

https://aesimtech.s3.us-east-1.amazonaws.com/artifacts/8MiAE6Rsr4POgq20ogQgkqsrVmv80Q2n/simba.exe
pip install aesim.simba==2023.10.17a1

@jcornau, you can check that it works and if it is the case you can close this issue.

closed 2023-10-19 by manu-aesim