aesim-tech/simba-project #368: Gear box model

AdriPWS, 2023-11-13:
That would be nice to build and add a Gear box model in SIMBA library for the next release?

Thanks,
KR,

manu-aesim, 2023-11-13:
I believe that it should be equivalent to an ideal transformer model, but we should verify this.

AdriPWS, 2023-11-13:
I agree with you Manu.

jcornau, 2025-03-18:
For doc
Simulink : 
https://fr.mathworks.com/help/simscape/ref/gearbox.html

Plecs: 

[image]

PSIM : 

[image]

jcornau, 2025-03-18:
Gearbox model added
Ideal model with gear ratio
W2/W1 = T1/T2 = g_ratio
if g_ratio < 0 --> shaft 1 and 2 are rotating in opposite direction