aesim-tech/simba-project #384: PMSM Output Torque Bug

manu-aesim, 2023-12-11:


imaryoko, 2024-01-08:
<Investigation note>
- The torque from the scope of the motor block shows reasonable value but the torque from the torque probe always shows unreasonably high value (e.g. 1e6 Nm) regardless of the motor model block types.
- From the theory, confirmed that the measured torque does not have to be equal to the torque from the machine scope. It's because the torque from the machine scope is the airgap torque, in the meantime, the measured torque is shaft torque.

imaryoko, 2024-01-08:
**Root cause**
This is a bug.

In motor blocks, the initial speed (speed at t<0) is defined at 0 rpm internally.
In the meantime, when connecting the constant speed load to a motor block, the motor speed does NOT ramp up gradually but rather starts at the specified speed from the first simulation step.
As a result, the motor shaft torque at the first step was always showing physically unreasonable high value.
E.g.:
T_ag = Jdw/dt + Bw + T_shaft
= J (w_now - w_prev)/(t_now - t_prev) + Bw +T_shaft

T_ag: Air gap torque (torque generated by the motor, scope value from a motor block)
T_shaft: Shaft torque ("torque generated by the motor" - "torque used by motor inertia", torque probe value)
J:Inertia,
B: friction
w: speed,
t: time

In this bug, the component of (w_now - w_prev)/(t_now - t_prev) was causing physically unreasonable answer at the first step.

**Solution**
Add an input field for the initial speed to motor blocks. (The same sense that the capacitance in SIMBA has a input filed for the initial voltage).

closed 2024-03-21 by manu-aesim