aesim-tech/simba-project #416: Controlled Speed Source

guillaumefontes, 2024-04-04:
As a user, I would like to get a controlled speed source to model loads that impose a speed. This speed can depend on a custom function (computed with control blocks or from a C-code), or can be read from a .csv file for example to apply a "speed profile".

manu-aesim, 2024-04-04:
This is quite simple to implement. Since this is a user request, we will prioritize this.

imaryoko, 2024-04-05:
Hi, I implemented controlled speed. Here's GUI.
[image]

manu-aesim, 2024-04-05:
Thanks Ryoko!

https://aesimtech.s3.us-east-1.amazonaws.com/artifacts/ybPnxOAweYqNEhwYgEtFLJ5ZijNVERX7/simba.exe
pip install aesim.simba==2024.4.5a1

closed 2024-04-05 by manu-aesim

AdriPWS, 2024-04-08:
Do you plan to add an "out" scope for this new block?
Actually only "in" scope is mentioned.

manu-aesim, 2024-04-08:
I don't understand. They can still use the "In" probe, isn't ?

AdriPWS, 2024-04-08:
ahah yes, they can but I assume MAYBE they will ask " why the output is not available"?
We can let like this and see later, no issue.