aesim-tech/simba-project #499: PID controller updates

manu-aesim, 2025-09-02:
**Continuous**
- Anti-windup support
- Saturation 
- Derivative filter
- Initial Integral Contribution 
- New scopes:
    - Proportional contribution 
    - Integral contribution
    - Derivative contribution
    - Output before saturation
    - Output after saturation
    
**Discrete**

jerome-secretgit, 2025-10-21:
Continuous

- [ ] Anti-windup support
- [ ] Saturation
- [ ] Derivative filter
- [ ] Initial Integral Contribution
- [ ] Integration Method
-     New scopes:
- [ ]         Proportional contribution
- [ ]         Integral contribution
- [ ]         Derivative contribution
- [ ]         Output before saturation
- [ ]         Output after saturation

jerome-secretgit, 2025-10-24:
[new simba results](https://github.com/user-attachments/files/23123278/new_chart.csv)
[old simba results](https://github.com/user-attachments/files/23123277/old_chart.csv)

Tests : 

- [x] compare with old release of Simba
- [ ] tests in simba.solver
- [ ] compare with PLECS